khoih-prog/AsyncHTTPRequest_Generic #2: HTTPS not supported?

julienaltieri, 2020-11-01:
ESP32 crashes when calling a "https://" url. This is 100% replicable on my ESP32 from the provided example: try changing the toronto.txt to an https address and the below issue happens. Since I needed to use this for REST API calls, SSL is required since I need to pass API keys. Here is what happens when I try to call GET on an https address:

`Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
`
After a bit of googling, I found this happens when there is some sort of null pointer access. 
After decoding the backtrace: 

`0x400d1a59: xbuf::write(char const*) at C:\Users\User\Documents\Arduino\libraries\AsyncHTTPRequest_Generic\src/utility/xbuf_Impl.h line 50
0x400d239a: AsyncHTTPRequest::_buildRequest() at C:\Users\User\Documents\Arduino\libraries\AsyncHTTPRequest_Generic\src/AsyncHTTPRequest_Impl_Generic.h line 527
0x400d24c9: AsyncHTTPRequest::send() at C:\Users\User\Documents\Arduino\libraries\AsyncHTTPRequest_Generic\src/AsyncHTTPRequest_Impl_Generic.h line 156
0x400d2ecb: sendRequest() at C:\Users\User\Desktop\RemoteJi/b_firebaseHandler.ino line 24`

khoih-prog, 2020-11-01:
The current version of [**AsyncHTTPRequest_Generic Library**](https://github.com/khoih-prog/AsyncHTTPRequest_Generic) is based on
1. [**AsyncTCP Library**](https://github.com/me-no-dev/AsyncTCP) and
2. [**asyncHTTPrequest Library**](https://github.com/boblemaire/asyncHTTPrequest) 

and all don't support HTTPS yet, AFAIK. I haven't tested any of them with HTTPS yet.

As this  [**AsyncHTTPRequest_Generic Library**](https://github.com/khoih-prog/AsyncHTTPRequest_Generic) is intended to support mainly other boards besided ESP32/ESP8266, I suggest you'd better post an issue in 

1. [**AsyncTCP Library**](https://github.com/me-no-dev/AsyncTCP) and
2. [**asyncHTTPrequest Library**](https://github.com/boblemaire/asyncHTTPrequest) 

Also check 

1. [**SSL support?]**(https://github.com/me-no-dev/AsyncTCP/issues/8)
2. [**mbed TLS client support, try 2**](https://github.com/me-no-dev/AsyncTCP/pull/48)
3  [**Can't make secure HTTP (HTTPS) request**](https://github.com/boblemaire/asyncHTTPrequest/issues/22) issue in [**asyncHTTPrequest Library**](https://github.com/boblemaire/asyncHTTPrequest).

> Sorry, the library only supports HTTP requests. HTTPS is not possible on the ESP8266 due to heap issues. On the ESP32, the underlying asyncTCP does not have support for TLS (HTTPS).

Please update if you found out solutions later.

closed 2020-11-01 by khoih-prog

khoih-prog, 2020-11-09:
@julienaltieri 

Can you post the simplified sketch to enable duplicating the issue.

I'll try to find out why and, if possible, add the **HTTPS** support to these Libraries, including this [**AsyncHTTPRequest_Generic**](https://github.com/khoih-prog/AsyncHTTPRequest_Generic).

I'm using **SSL** successfully in [**Blynk_Async_WM Library**](https://github.com/khoih-prog/Blynk_Async_WM) to connect to SSL Local Blynk Server. Also check [**Using Private Blynk server in SSL mode**](https://github.com/khoih-prog/Blynk_Async_WM/issues/1) and see **Async SSL** is working, using [**ESPAsyncWebServer**](https://github.com/me-no-dev/ESPAsyncWebServer).

But certainly, you have to store **Server's SSL Public Certificates** locally in the ESP32/ESP8266 boards. Did you?

Good Luck with your project.

julienaltieri, 2020-11-10:
Simplest example is from this [sketch](https://github.com/boblemaire/asyncHTTPrequest/blob/master/sample/sample.cpp) 
When I replace http by https, I can trigger the error. 
Note that I get no issue accessing this same https:// address using the classic HTTPClient for ESP32 as described in [this post](https://techtutorialsx.com/2017/05/20/esp32-http-post-requests/)
(note the capital HTTP in HTTPClient which is different from the traditional Arduino library)

In other words, if SSL and asynchronousness are fundamentally independent (as I imagine), there should be a way to support async https on ESP32 if we could marry the best of both worlds. (although that might not be easy)

Thank you for following up on this!