khoih-prog/AsyncHTTPRequest_Generic #3: AsyncHTTPRequest_Generic with. AsyncMqttClient, ESP8266 reboots

tokra, 2020-11-07:
I tried to use AsyncHTTP req when i receive message from MQTT topic (using AsyncMqttClient), with code:
```c
AsyncHTTPRequest request;
  request.onReadyStateChange([](void* optParm, AsyncHTTPRequest* request, int readyState) {
    Serial.println(readyState);
    if (readyState == readyStateDone) {
      Serial.println("\n**************************************");
      Serial.println(request->responseText());
      Serial.println("**************************************");
    }
  });
  request.open("GET", "http://worldtimeapi.org/api/timezone/Europe/Bratislava");
  request.send();
```
inside `void onMqttMessage(char* topic, char* payload, AsyncMqttClientMessageProperties properties, size_t len, size_t index, size_t total) ` of AsyncMQTT client.
When i send message to mqtt topic and received on ESP, board reboots after  while it reaches AsyncHttp client
```log
21:30:34.913 -> MQTT | Receive handler:
21:30:34.913 -> /ping
21:30:34.913 -> {"from":"tower","msg":"are u alive?"}aop⸮
21:30:42.180 -> 
21:30:42.180 ->  ets Jan  8 2013,rst cause:4, boot mode:(3,6)
21:30:42.180 -> 
21:30:42.180 -> wdt reset
21:30:42.180 -> load 0x4010f000, len 1384, room 16 
21:30:42.180 -> tail 8
21:30:42.180 -> chksum 0x2d
21:30:42.180 -> csum 0x2d
21:30:42.180 -> vbc204a9b
21:30:42.180 -> ~ld
```

Is this client compatible and can it be used alongside with [AsyncMqttClient](https://github.com/marvinroger/async-mqtt-client) ?

khoih-prog, 2020-11-09:
@tokra 

Thanks for using the library.

In order to trace / fix the issue and also save our time, you have to post the **minimum code that anybody can duplicate the issue.**

Without the duplication, it's not possible to figure out and/or fix any issue.

Some preliminary suggestions:

1. The [**AsyncMQTTClient**](https://github.com/marvinroger/async-mqtt-client) has many problems, as pointed out in [**PangolinMQTT**](https://github.com/philbowles/PangolinMQTT). I suggest you give that [**PangolinMQTT**](https://github.com/philbowles/PangolinMQTT)  library a try to see if OK for you.

2. This [**AsyncHTTPRequest_Generic**](https://github.com/khoih-prog/AsyncHTTPRequest_Generic) is modified from the original [**asyncHTTPrequest**](https://github.com/boblemaire/asyncHTTPrequest) for other boards besides ESP8266/ESP32. As you're using **ESP8266**, it's better that you seek the support there by open an issue in [**asyncHTTPrequest Issue**](https://github.com/boblemaire/asyncHTTPrequest/issues) as many more experienced ESP users / experts there can help.

I'm closing the issue now, until something new arises.

Good luck,

closed 2020-11-09 by khoih-prog

khoih-prog, 2020-11-09:
Hi @tokra 

I've just updated the library to new [**Version v1.0.2**](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/releases/tag/v1.0.2) to fix  potential issue with **Mutex and delete function.**

Can you retest to see if your issue is fixed.

---

### Releases v1.0.2

1. Make **Mutex Lock and delete more reliable and error-proof to prevent random crash.**

khoih-prog, 2020-11-09:
Hopefully this [**Random crashes in asyncHTTPrequest::_onDisconnect**](https://github.com/boblemaire/asyncHTTPrequest/issues/19) is related to the Mutex and Delete issue here on ESP32.

tokra, 2020-11-09:
Hey, i found the issue in my code. thanks anyway.