khoih-prog/AsyncHTTPRequest_Generic #4: Send PUT request

babca, 2020-12-15:
Sending `request.open("PUT", "http://domain.com/api/endpoint/");` currently raises Guru Meditation Error.

How to send PUT request? It is technically no different from POST, except its name.

khoih-prog, 2020-12-15:
Thanks for your interest in the library.

The **PUT** is not currently supported, only **POST** and **GET.** You can rewrite you code to use POST instead of PUT.

If you'd like us to consider to add the **enhancement** in the near future, at first, could you read the document [**How to create a Minimal, Reproducible Example**](https://stackoverflow.com/help/minimal-reproducible-example) as well as [**Reporting Bugs**](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/blob/master/CONTRIBUTING.md#reporting-bugs)

Then post the reproducible example to save us some time to reproduce the issue and test the new enhancement.

I'll close the issue and reopen when you post the reproducible code as guidelined.

closed 2020-12-15 by khoih-prog

khoih-prog, 2020-12-23:
Please check the new [**AsyncHTTPRequest_Generic Library v.1.1.0**](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/releases/tag/v1.1.0) which has just been released.

Please test and verify to see if there is any issue / bug.

Thanks and Regards,

KH

---

### Releases v1.1.0

1. Add HTTP **PUT, PATCH, DELETE and HEAD methods**. Check [**Add support for sending PUT, PATCH, DELETE reques**t](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/issues/5)
2. Add Table of Contents
3. Add Version String