khoih-prog/AsyncHTTPRequest_Generic #7: Unable to resolve Multiple Definitions issue

joegatling, 2020-12-29:
Hello, I am trying to get this library to compile, but I am encountering the multiple definitions linker issue that is mentioned in the documentation. I am having a bit of trouble with the specified workaround.

As far as I can follow, the instructions are:
1. Rename the **src** folder containing nothing but .h files to **src_h**
2. Rename the **src_cpp** folder to **src**

The issue is that the library seems to already have a folder called **src_h**, which seems to be a duplicate of **src**.

I have tried simply deleting the src folder and renaming src_cpp, but that results in the following error:
`fatal error: AsyncHTTPRequest_Generic.h: No such file or directory
 #include <AsyncHTTPRequest_Generic.h> 
                                       ^`

Attached is a screenshot of the folders I have.
[image]

Any advice on how to fix this would be greatly appreciated!

khoih-prog, 2020-12-29:
It seems you've done something not fully correct as the error states that you have **no file `AsyncHTTPRequest_Generic.h` at all in `src` directory.**

Make sure the only remaining `src` directory contains these files 

[image]

> The issue is that the library seems to already have a folder called src_h, which seems to be a duplicate of src.

You can follows these steps (so obvious that I even didn't mentioned)

1. Delete the `src_h` directory.
2. Detele `src` directory
3. Rename `src_cpp` -> `src`

Remember also to close then reopen PIO to refresh.

closed 2020-12-29 by khoih-prog

joegatling, 2021-01-02:
Oh great! Thanks for clearing that up for me.

regards
-Joe