khoih-prog/AsyncHTTPRequest_Generic #8: Clarification on the license?

mairas, 2021-01-16:
This library is obviously a fork of https://github.com/boblemaire/asyncHTTPrequest. While I much prefer the MIT license used in this project, it is not compatible with GPLv3 used in the original repository. According to the GPL licenses, any derivatives need to be licensed with the same license.

Could this issue be clarified? Did the authors of the original library grant a permission for relicensing? If not, I believe this repository should be relicensed with GPLv3.

khoih-prog, 2021-01-16:
I'm sorry I didn't spent time to look at the license and see the differences. It doesn't matter to me if it's GPLv3 or MIT, as long as users can use the library easily.

If there is issue, I'll convert back to GPLv3 without any problem at all.

Thanks for pointing out.

khoih-prog, 2021-01-16:
Anyway, I already change the license to GPLv3 to avoid any issue. Thanks.

closed 2021-01-16 by khoih-prog