khoih-prog/AsyncHTTPRequest_Generic #10: Request Large Data is an issue?

HackEduca, 2021-02-03:
Hi, Congratulation for this library.
I would like to check if is there a way to increase the buffer to receive large data (considering that this is the cause ;o) )
Currently I am running the example :  **AsyncHTTPRequest_ESP.ino**

I have replaced the following data:
From:
`requestOpenResult = request.open("GET", "http://worldtimeapi.org/api/timezone/America/Toronto.txt");`

To:
`requestOpenResult = request.open("GET", "http://api.openweathermap.org/data/2.5/weather?q=Peruibe,br&appid=36768002e8b918d8a071d6c331ab713c&q");   `

This works fine, but when I try to get a full json file in this one:
`requestOpenResult = request.open("GET", "https://api.openweathermap.org/data/2.5/onecall?lat=-24.32&lon=-46.9983&exclude={part}&appid=36768002e8b918d8a071d6c331ab713c&q");`

No data is comming and I it's triggered the message from that part of the code.
`Serial.println("Can't send bad request");`

Thanks!

Edson

khoih-prog, 2021-02-03:
Thanks for your interest and nice words about the library.

As you know from the library name [**AsyncHTTPRequest_Generic**](https://github.com/khoih-prog/AsyncHTTPRequest_Generic), it supports only **HTTP** request. But you passed a **HTTPS** request which was rejected.

Please change the request to HTTP as follows

```
requestOpenResult = request.open("GET", "http://api.openweathermap.org/data/2.5/onecall?lat=-24.32&lon=-46.9983&exclude={part}&appid=36768002e8b918d8a071d6c331ab713c&q");
```

then everything will be OK.

closed 2021-02-03 by khoih-prog

HackEduca, 2021-02-04:
Thanks for your prompt reply.
I am using a node MCU and I got and empty space as a result this time
Testing with:
`requestOpenResult = request.open("GET", "http://api.openweathermap.org/data/2.5/onecall?lat=-24.32&lon=-46.9983&exclude={part}&appid=36768002e8b918d8a071d6c331ab713c&q");`
[image]

If I try the smaller content I got it (which is fine)

Testing with:
`requestOpenResult = request.open("GET", "http://api.openweathermap.org/data/2.5/weather?q=Peruibe,br&appid=36768002e8b918d8a071d6c331ab713c&q");`
[image]

Thanks!

Edson

khoih-prog, 2021-02-04:
The issue is either there is not enough heap memory or a bug in the library to handle too big response String (more than 20K).

The current solution is

1. You have to divide the query into **several smaller ones** (current,minutely,hourly,daily,alerts) to handle. Even so, the **hourly response is still too large to handle**, so it's better to divide the hourly query further.

Check the new example [**AsyncHTTPMultiRequests_ESP**](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/tree/master/examples/AsyncHTTPMultiRequests_ESP) written to divide the query into multiple ones.

I don't think I have time now to spend on this special issue as you can have a dividing work-around.

2. As this is **ESP-related issue** as well, it's better to post the issue to the original [**asyncHTTPrequest library issue**](https://github.com/boblemaire/asyncHTTPrequest/issues) where the author Bob Lemaire, I believe, is much more experienced  in solving this issue.

So I still leave this issue closed now and please post if you have new solution.

khoih-prog, 2021-02-04:
The Terminal output when running [**AsyncHTTPMultiRequests_ESP**](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/tree/master/examples/AsyncHTTPMultiRequests_ESP) 

```
HTTP WebServer is @ IP : 192.168.2.82

***************Current***************
{"lat":-24.32,"lon":-46.9983,"timezone":"America/Sao_Paulo","timezone_offset":-10800,"current":{"dt":1612412144,"sunrise":1612428484,"sunset":1612475747,"temp":297.71,"feels_like":301.72,"pressure":1011,"humidity":88,"dew_point":295.59,"uvi":0,"clouds":100,"visibility":10000,"wind_speed":1.32,"wind_deg":357,"weather":[{"id":804,"main":"Clouds","description":"overcast clouds","icon":"04n"}]}}
**************************************
H
***************Minutely***************
{"lat":-24.32,"lon":-46.9983,"timezone":"America/Sao_Paulo","timezone_offset":-10800,"minutely":[{"dt":1612412160,"precipitation":0},{"dt":1612412220,"precipitation":0},{"dt":1612412280,"precipitation":0},{"dt":1612412340,"precipitation":0},{"dt":1612412400,"precipitation":0},{"dt":1612412460,"precipitation":0},{"dt":1612412520,"precipitation":0},{"dt":1612412580,"precipitation":0},{"dt":1612412640,"precipitation":0},{"dt":1612412700,"precipitation":0},{"dt":1612412760,"precipitation":0},{"dt":1612412820,"precipitation":0},{"dt":1612412880,"precipitation":0},{"dt":1612412940,"precipitation":0},{"dt":1612413000,"precipitation":0},{"dt":1612413060,"precipitation":0},{"dt":1612413120,"precipitation":0},{"dt":1612413180,"precipitation":0},{"dt":1612413240,"precipitation":0},{"dt":1612413300,"precipitation":0},{"dt":1612413360,"precipitation":0},{"dt":1612413420,"precipitation":0},{"dt":1612413480,"precipitation":0},{"dt":1612413540,"precipitation":0},{"dt":1612413600,"precipitation":0},{"dt":1612413660,"precipitation":0},{"dt":1612413720,"precipitation":0},{"dt":1612413780,"precipitation":0},{"dt":1612413840,"precipitation":0},{"dt":1612413900,"precipitation":0},{"dt":1612413960,"precipitation":0},{"dt":1612414020,"precipitation":0},{"dt":1612414080,"precipitation":0},{"dt":1612414140,"precipitation":0},{"dt":1612414200,"precipitation":0},{"dt":1612414260,"precipitation":0},{"dt":1612414320,"precipitation":0},{"dt":1612414380,"precipitation":0},{"dt":1612414440,"precipitation":0},{"dt":1612414500,"precipitation":0},{"dt":1612414560,"precipitation":0},{"dt":1612414620,"precipitation":0},{"dt":1612414680,"precipitation":0},{"dt":1612414740,"precipitation":0},{"dt":1612414800,"precipitation":0},{"dt":1612414860,"precipitation":0},{"dt":1612414920,"precipitation":0},{"dt":1612414980,"precipitation":0},{"dt":1612415040,"precipitation":0},{"dt":1612415100,"precipitation":0},{"dt":1612415160,"precipitation":0},{"dt":1612415220,"precipitation":0},{"dt":1612415280,"precipitation":0},{"dt":1612415340,"precipitation":0},{"dt":1612415400,"precipitation":0},{"dt":1612415460,"precipitation":0},{"dt":1612415520,"precipitation":0},{"dt":1612415580,"precipitation":0},{"dt":1612415640,"precipitation":0},{"dt":1612415700,"precipitation":0},{"dt":1612415760,"precipitation":0}]}
**************************************
H
***************Daily***************
{"lat":-24.32,"lon":-46.9983,"timezone":"America/Sao_Paulo","timezone_offset":-10800,"daily":[{"dt":1612450800,"sunrise":1612428484,"sunset":1612475747,"temp":{"day":307.68,"min":296.97,"max":308.75,"night":299,"eve":303.79,"morn":296.97},"feels_like":{"day":311.49,"night":301.16,"eve":307.77,"morn":300.49},"pressure":1008,"humidity":47,"dew_point":294.96,"wind_speed":0.93,"wind_deg":353,"weather":[{"id":501,"main":"Rain","description":"moderate rain","icon":"10d"}],"clouds":47,"pop":0.96,"rain":2.79,"uvi":15.33},{"dt":1612537200,"sunrise":1612514927,"sunset":1612562116,"temp":{"day":299.25,"min":296.76,"max":300.13,"night":296.76,"eve":298.25,"morn":297.21},"feels_like":{"day":302.3,"night":299.49,"eve":300.98,"morn":300.65},"pressure":1007,"humidity":77,"dew_point":294.96,"wind_speed":2.16,"wind_deg":162,"weather":[{"id":501,"main":"Rain","description":"moderate rain","icon":"10d"}],"clouds":96,"pop":1,"rain":9.37,"uvi":7.72},{"dt":1612623600,"sunrise":1612601368,"sunset":1612648485,"temp":{"day":297.87,"min":295.8,"max":298.07,"night":295.8,"eve":296.4,"morn":296.09},"feels_like":{"day":300.39,"night":297.49,"eve":298.26,"morn":298.75},"pressure":1010,"humidity":83,"dew_point":294.79,"wind_speed":2.84,"wind_deg":130,"weather":[{"id":501,"main":"Rain","description":"moderate rain","icon":"10d"}],"clouds":78,"pop":0.83,"rain":12.85,"uvi":10.28},{"dt":1612710000,"sunrise":1612687809,"sunset":1612734852,"temp":{"day":299.97,"min":294.23,"max":299.97,"night":295.42,"eve":297.96,"morn":294.23},"feels_like":{"day":300.18,"night":296.37,"eve":298.39,"morn":295.74},"pressure":1013,"humidity":57,"dew_point":290.86,"wind_speed":3.43,"wind_deg":159,"weather":[{"id":500,"main":"Rain","description":"light rain","icon":"10d"}],"clouds":37,"pop":0.83,"rain":3.06,"uvi":13.6},{"dt":1612796400,"sunrise":1612774250,"sunset":1612821218,"temp":{"day":300.37,"min":294.91,"max":300.37,"night":295.95,"eve":298.22,"morn":295.06},"feels_like":{"day":300.71,"night":297.04,"eve":298.9,"morn":297.22},"pressure":1014,"humidity":57,"dew_point":291.16,"wind_speed":3.47,"wind_deg":147,"weather":[{"id":500,"main":"Rain","description":"light rain","icon":"10d"}],"clouds":12,"pop":0.32,"rain":1.25,"uvi":5.63},{"dt":1612882800,"sunrise":1612860690,"sunset":1612907582,"temp":{"day":300.92,"min":294.21,"max":300.92,"night":295.84,"eve":297.77,"morn":294.21},"feels_like":{"day":301.27,"night":29 [... cut ...]
{"lat":-24.32,"lon":-46.9983,"timezone":"America/Sao_Paulo","timezone_offset":-10800,"alerts":[{"sender_name":"Instituto Nacional de Meteorologia","event":"Tempestade","start":1612364400,"end":1612447200,"description":"INMET publica aviso iniciando em: 03/02/2021 12:00. Chuva entre 20 e 30 mm/h ou até 50 mm/dia, ventos intensos (40-60 km/h), e queda de granizo. Baixo risco de corte de energia elétrica, estragos em plantações, queda de galhos de árvores e de alagamentos."}]}
**************************************

```

HackEduca, 2021-02-04:
Will, try thanks a lot.