khoih-prog/AsyncHTTPRequest_Generic #11: compatibility with ESPAsyncWebServer

baddwarf, 2021-02-11:
Hi,

Please forgive me if it's an easy question. I'm a casual programmer.

I'm trying to use your library and ESPAsyncWebServer library [https://github.com/me-no-dev/ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer). I've got a compilation error regarding _lock value (which is defined in both project).

I'm trying to understand what this variable is used for in your library but I can't find out. Is there a way to rename/bypass this and getting the two library working together?

khoih-prog, 2021-02-11:
You can try the simple solution as follows to see if it's working for you

```
#include <AsyncHTTPRequest_Generic.h>        // https://github.com/khoih-prog/AsyncHTTPRequest_Generic
AsyncHTTPRequest request;

...

#undef _lock                                 // to undefine _lock in AsyncHTTPRequest_Generic.h to avoid conflict

#include <ESPAsyncWebServer.h>
```

If not, you can temporarily modify either `ESPAsyncWebServer.h` or `AsyncHTTPRequest_Generic.h` and change from `_lock` to `_localLock` or whatever unique.

In the future release of [**AsyncHTTPRequest_Generic**](https://github.com/khoih-prog/AsyncHTTPRequest_Generic), I'll try to use namespace to avoid name-pollution issue.

closed 2021-02-11 by khoih-prog

khoih-prog, 2021-02-11:
I'll release a new version today to fix that conflict. Hold on if you still have some issue.

baddwarf, 2021-02-11:
Many thanks. I'll check that ASAP.

khoih-prog, 2021-02-11:
Hi @baddwarf 

The new release has been published. You contribution has been noted in [**Contributions and Thanks**](https://github.com/khoih-prog/AsyncHTTPRequest_Generic#contributions-and-thanks). Don't be humble anymore, we, Library Writers, need many **fresh eyes and minds** to test, use and point out many of our mistakes and omissions.

Please test and report if there is any further issue.

---
---
### Releases v1.1.2

1. Rename _lock and _unlock to avoid conflict with [**ESP32/ESP8266 AsyncWebServer**](https://github.com/me-no-dev/ESPAsyncWebServer) library. Check [**compatibility with ESPAsyncWebServer #11**](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/issues/11)
2. Fix compiler warnings.