khoih-prog/AsyncHTTPRequest_Generic #13: 'Connection' header expects 'disconnect' instead 'close' ?

spdi, 2021-02-21:
Hi,
I'm using your library to connect to me-no-dev ESPAsyncWebServer to make a POST request with body.
I had some troubles to make it work beacuse ESPAsyncWebServer accepted only first call but second was not sent because requestState returns 1.
Reading the source code I found that the Connection header is compared to have 'disconnect' value to disconnect TCP connection. So I fixed it to 'close' string and it started to work.
As https://tools.ietf.org/html/rfc2616#page-117 Connection should have 'close' value.
Could you explain why do you use 'disconnect'?
If this is bug could you fix it?

fixed line by me:
`if (connectionHdr && (strcasecmp_P(connectionHdr, PSTR("close")) == 0))`

Second discovered problem is debugging received data.
Data debug does not pass buffer data length to debug macro and data Vbuf is not null terminated so I read on screen the received body response and some garbage.

BTW I like your library because it support POST and other not GET methods. Thank you very much.

khoih-prog, 2021-02-22:
HI @spdi 

Thanks for pointing out the bug. You're correct that the response must be

```
Connection: close
```
for declaring non-persistence connections per [**Compatibility with HTTP/1.0 Persistent Connections**](https://tools.ietf.org/html/rfc2616#section-19.6.2)

> Persistent connections are the default for HTTP/1.1 messages; we introduce a new keyword (Connection: close) for declaring non-persistence.

I'll correct and publish a new release to fix.

---

> Second discovered problem is debugging received data.
> Data debug does not pass buffer data length to debug macro and data Vbuf is not null terminated so I read on screen the received body response and some garbage.

I'd appreciate it if you can create a PR to save me some time to investigate the `non null-terminated` bug. If not, can you provide a simple code to reproduce the issue so that I can trace and fix the `non null-terminated` bug.

Best Regards,

khoih-prog, 2021-02-25:
New release to fix non-persistent Connection header bug

---

### Releases v1.1.3

1. Fix non-persistent Connection header bug. Check [**'Connection' header expects 'disconnect' instead 'close' ? #13**](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/issues/13)
2. Add ESP32-S2 support
3. Tested with [**Latest ESP32 Core 1.0.5**](https://github.com/espressif/arduino-esp32) for ESP32-based boards.

closed 2021-02-25 by khoih-prog