khoih-prog/AsyncHTTPRequest_Generic #14: ESP8266 issue platformio

mk3d, 2021-04-01:
Hi, thanks for your work!
I have an issue trying to compile for ESP8266

This error appear:
.pio/libdeps/nodemcuv2/STM32Ethernet/src/utility/stm32_eth.h:42:23: fatal error: stm32_def.h: No such file or directory

lib_deps = 
	khoih.prog/AsyncHTTPRequest_Generic@^1.1.5

In main.cpp
AsyncHTTPRequest request;

Any idea? Why ESP32 desp for ESP8266?
Many thanks and best regards.

khoih-prog, 2021-04-01:
It seems that you got into the same issue as in [**Missing stm32_def.h for Teensy 4.1** #14](https://github.com/khoih-prog/WebSockets2_Generic/issues/14) of [**WebSockets2_Generic** library](https://github.com/khoih-prog/WebSockets2_Generic).

To fix, please replace [library.json Lines 45 to 66](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/blob/master/library.json#L45-L66)

```
    {
      "owner": "stm32duino",
      "name": "STM32duino LwIP",
      "version": ">=2.1.2",
      "platforms": "ststm32"
    },
    {
      "owner": "stm32duino",
      "name": "STM32Ethernet",
      "version": ">=1.2.0",
      "platforms": "ststm32"
    },
    {
      "owner": "lorol",
      "name": "LittleFS_esp32",
      "version": ">=1.0.5",
      "platforms": ["espressif32"]
    },
    {
      "name": "STM32AsyncTCP",
      "version": "https://github.com/khoih-prog/STM32AsyncTCP"
    }
```

with

```
    {
      "owner": "lorol",
      "name": "LittleFS_esp32",
      "version": ">=1.0.5",
      "platforms": ["espressif32"]
    }
```

Those lines were added as **precaution (more than just enough, covering all platforms / libraries)** so that PIO users can easily select which are necessary.

If that deletion won't solve your issue, I suggest you post on PIO forum to ask for help from many experienced experts there.

If you find out and solve the issue, please giving back and help other users by posting the fix here.

Good Luck,

closed 2021-04-01 by khoih-prog

jernkuan, 2021-09-16:
Adding 

lib_compat_mode = strict

to platformio.ini under your environment works for me.

khoih-prog, 2021-09-16:
@jernkuan 

Thanks for your sharing. I'll update `platform.ini` for all of my libs.