khoih-prog/AsyncHTTPRequest_Generic #15: POSTing data appears to just append to url

andyb2000, 2021-04-08:
### Describe the bug

Using method of POST based on example appears to just still append variables to URL.
Using the examples:
    String server_url="http://myserver.domain/posturl.php";
    String paramdata = "variable1=value&variable2=value";
    requestOpenResult = request.open("POST", (server_url + "?" + paramdata).c_str());

### Steps to Reproduce

Above code, checking the server it shows the data being pushed on the url:
x.x.x.x - - [08/Apr/2021:13:47:27 +0100] "POST /posturl.php?variable1=value&variable2=value" HTTP/1.1" 200 147 "-" "-"

### Expected behavior

Should see just the POST itself and no parameters on the url (a-la-get style):
x.x.x.x - - [08/Apr/2021:13:47:27 +0100] "POST /posturl.php" HTTP/1.1" 200 147 "-" "-"

### Information

Please ensure to specify the following:

* Arduino IDE version1.8.12
* ESP8266
* AsyncHTTPRequest_Generic version 1.1.5

I'm unsure if this is an error in the examples or if it's not possible to POST in this way.

khoih-prog, 2021-04-08:
Hi @andyb2000 

The library code just send whatever you ask it to send. With

```
requestOpenResult = request.open("POST", (server_url + "?" + paramdata).c_str());
```

The server will receive

```
server_url + "?" + paramdata ===> /posturl.php?variable1=value&variable2=value
```

If you need the server to receive just `/posturl.php`, send

```
requestOpenResult = request.open("POST", server_url.c_str());
```

khoih-prog, 2021-04-08:
Hi @andyb2000 

Sorry for previous hasty and wrong reply, if you received.

The current [**AsyncDweetPost_STM32** example](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/tree/master/examples/AsyncDweetPost_STM32) is written for normal (not **php**) WebServer, expecting the POST request in the following format

```
POST /varPost?variable1=value&variable2=value
```
or

```
POST /dweet/for/pinA0-Read?sensorValue=100
```

while for **php WebServer**, the following kind of data is expected

```
POST /posturl.php  HTTP/1.1
Host: myserver.domain
variable1=value&variable2=value
```

So you have to use send() functions, with **paramdata** for POST, as follows:

```
    requestOpenResult = request.open("POST", server_url.c_str());

    if (requestOpenResult)
    {
      // Only send() if open() returns true, or crash
      request.send(paramdata);    // POST data here
    }
    else
    {
      Serial.println("Can't send bad request");
    }
```

You can see the funtion prototypes here

```
bool        send(String body);                                      // Send the request (POST)
bool        send(const char* body);                                 // Send the request (POST)
bool        send(const uint8_t* buffer, size_t len);                // Send the request (POST) (binary data?)
bool        send(xbuf* body, size_t len);                           // Send the request (POST) data in an xbuf
```

closed 2021-04-08 by khoih-prog

andyb2000, 2021-04-09:
Thanks @khoih-prog appreciate you solving and no problem on the early reply!

That answers what I understood, thank you :-)

andyb2000, 2021-04-09:
For anyone else coming to this later, I spent a lot of time debugging why this still wouldn't work. The following worked:

    requestOpenResult = request.open("POST", server.c_str());
    request.setReqHeader("X-Content-Type", "application/x-www-form-urlencoded");
    request.setReqHeader("Content-Type", "application/x-www-form-urlencoded");
    request.setReqHeader("Host", "myhost.com");
    
    if (requestOpenResult)
    {
      request.send("param1=test\r\n");
   }

khoih-prog, 2021-04-09:
Hi @andyb2000 

It's good that you quickly and brightly solve the problem.
If possible, could you please post your simplest **php** code so that other users as well as I can use and test.
I didn't post the fully correct solution because I couldn't duplicate the issue, without the same php.
I'll create and publish several new examples using **POST + php**, certainly with your credits.

Cheers,

andyb2000, 2021-04-12:
No problem!
The PHP code is simple, I just log the variables being received to a temporary file. This will dump all variables received by the php script to the file so you can review/analyse.

`<?php
$req_dump = print_r($_REQUEST, TRUE);
$fp = fopen('/tmp/request.log', 'a');
fwrite($fp, $req_dump);
fclose($fp);
exit;
?>`

Thanks!