khoih-prog/AsyncHTTPRequest_Generic #16: Rapid http get polling causes crash when host disconnected

TLS1000, 2021-05-07:
Hi, I'm polling an energy meter with your library, and it works fine, however when I disconnect the host device the timeout mechanism doesn't seem te work and the ESP32 crashes

I used the 'assynchttprequestESP' with my host (energy meter) example, changed the request frequency to 2s, and tried to set the timeout to 1s by '#define DEFAULT_RX_TIMEOUT 1                    // Seconds for timeout '

Timeout doesn't seem to do it's job, and also troubling is that it goes through the callbackhandler at some time before crashing and writes false values to my variables.

Am I missing something?

khoih-prog, 2021-05-07:
Thanks for using the library.

This seems to be an interesting issue. But I'm sorry I can't spend time on this until you follow the instructions when posting a new [**Issue: Bug report**](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/issues/new?assignees=&labels=&template=bug_report.md&title=).

You have to provide the [**Minimal, Reproducible Example**](https://stackoverflow.com/help/minimal-reproducible-example) so that anybody can reproduce the result and find the bug, if exists.

I'm closing the issue now, and reopen whenever you prove that this is caused by a bug of this library.

Good Luck,

closed 2021-05-07 by khoih-prog

andrewk123, 2021-10-31:
A follow up to this I am seeing the same issue.

Bug report is as follows:

### Describe the bug

Multiple requests to an internal NodeJS server when the server is disconnected causes a crash.

### Steps to Reproduce

Use the example "AsyncHTTPRequest_ESP.ino"

Only changes in this sketch are as follows:
Line 10 #define HEARTBEAT_INTERVAL        10
Line 49 #define _ASYNC_HTTP_LOGLEVEL_  4
Line 59 and 60 add SSID and Password
Line 102 change to requestOpenResult = request.open("GET", "http://192.168.59.188/ping.htm");

### Expected behavior

When the target is disconnected an HTTP error code should be returned after a very short timeout.

### Actual behavior

When http://192.168.59.188/ping.htm is available everything works as expected (the ping page just generates an OK) which is printed by the Sketch.  However if you connect the ethernet cable.  The first request receives no response (as expected). After several requests (typically 3) the ESP restarts,

Please ensure to specify the following:

* Arduino IDE version (e.g. 1.8.13) or Platform.io version 
Arduino IDE 1.8.13
* `ESP8266`,`ESP32` or `STM32` Core Version (e.g. ESP8266 core v2.7.4, ESP32 v1.0.5 or STM32 v1.9.0)
ESP8266 Core 3.0.2

Here is the debug output:

AsyncHTTPRequest @ IP : 192.168.1.23
[AHTTP] open( GET , url = http://192.168.1.21/ping.htm
[AHTTP] _parseURL(): scheme+host HTTP:// 192.168.1.21
[AHTTP] _parseURL(): port+path+query 80 /ping.htm 
[AHTTP] open: conneting to hostname = 192.168.1.21:80
[AHTTP] _connect()
[AHTTP] send()
[AHTTP] _buildRequest()
[AHTTP] _HTTPmethod = 0
[AHTTP] GET  /ping.htm   HTTP/1.1

[AHTTP] host : 192.168.1.21:80 

[AHTTP] _send(), _request->available = 50
[AHTTP] *can't send
[AHTTP] open( GET , url = http://192.168.1.21/ping.htm
[AHTTP] _parseURL(): scheme+host HTTP:// 192.168.1.21
[AHTTP] _parseURL(): port+path+query 80 /ping.htm 
[AHTTP] open: conneting to hostname = 192.168.1.21:80
[AHTTP] _connect()
[AHTTP] send()
[AHTTP] _buildRequest()
[AHTTP] _HTTPmethod = 0
[AHTTP] GET  /ping.htm   HTTP/1.1

[AHTTP] host : 192.168.1.21:80 

[AHTTP] _send(), _request->available = 50
[AHTTP] *can't send
H[AHTTP] open( GET , url = http://192.168.1.21/ping.htm
[AHTTP] _parseURL(): scheme+host HTTP:// 192.168.1.21
[AHTTP] _parseURL(): port+path+query 80 /ping.htm 
[AHTTP] open: conneting to hostname = 192.168.1.21:80
[AHTTP] _connect()
[AHTTP] send()
[AHTTP] _buildRequest()
[AHTTP] _HTTPmethod = 0
[AHTTP] GET  /ping.htm   HTTP/1.1

[AHTTP] host : 192.168.1.21:80 

[AHTTP] _send(), _request->available = 50
[AHTTP] *can't send
H[AHTTP] open( GET , url = http://192.168.1.21/ping.htm
[AHTTP] _parseURL(): scheme+host HTTP:// 192.168.1.21
[AHTTP] _parseURL(): port+path+query 80 /ping.htm 
[AHTTP] open: conneting to hostname = 192.168.1.21:80
[AHTTP] _connect()
[AHTTP] send()
[AHTTP] _buildRequest()
[AHTTP] _HTTPmethod = 0
[AHTTP] GET  /ping.htm   HTTP/1.1

[AHTTP] host : 192.168.1.21:80 

[AHTTP] _send(), _request->available = 50
[AHTTP] *can't send
H[AHTTP] _onError handler error = -13
[AHTTP] 
_onDisconnect handler
[AHTTP] _setReadyState : 4

**************************************
[AHTTP] responseText()
[AHTTP] responseText() no data

**************************************

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Note:  It takes approximately 30 seconds from the first [AHTTP] *can't send till the _onDisconnect handler fires.  The reset happens at the time the next request would happen after the _onDisconnect handler fires.

Thanks for your help