khoih-prog/AsyncHTTPRequest_Generic #22: Http GET polling causes crash when host disconnected

andrewk123, 2021-10-31:
A follow up to this I am seeing the same issue.

Bug report is as follows:

### Describe the bug

Multiple requests to an internal NodeJS server when the server is disconnected causes a crash.

### Steps to Reproduce

Use the example "AsyncHTTPRequest_ESP.ino"

Only changes in this sketch are as follows:
Line 10 #define HEARTBEAT_INTERVAL        10
Line 49 #define _ASYNC_HTTP_LOGLEVEL_  4
Line 59 and 60 add SSID and Password
Line 102 change to requestOpenResult = request.open("GET", "http://192.168.59.188/ping.htm");

### Expected behavior

When the target is disconnected an HTTP error code should be returned after a very short timeout.

### Actual behavior

When http://192.168.59.188/ping.htm is available everything works as expected (the ping page just generates an OK) which is printed by the Sketch.  However if you connect the ethernet cable.  The first request receives no response (as expected). After several requests (typically 3) the ESP restarts,

Please ensure to specify the following:

* Arduino IDE version (e.g. 1.8.13) or Platform.io version 
Arduino IDE 1.8.13
* `ESP8266`,`ESP32` or `STM32` Core Version (e.g. ESP8266 core v2.7.4, ESP32 v1.0.5 or STM32 v1.9.0)
ESP8266 Core 3.0.2

Here is the debug output:

AsyncHTTPRequest @ IP : 192.168.1.23
[AHTTP] open( GET , url = http://192.168.1.21/ping.htm
[AHTTP] _parseURL(): scheme+host HTTP:// 192.168.1.21
[AHTTP] _parseURL(): port+path+query 80 /ping.htm 
[AHTTP] open: conneting to hostname = 192.168.1.21:80
[AHTTP] _connect()
[AHTTP] send()
[AHTTP] _buildRequest()
[AHTTP] _HTTPmethod = 0
[AHTTP] GET  /ping.htm   HTTP/1.1

[AHTTP] host : 192.168.1.21:80 

[AHTTP] _send(), _request->available = 50
[AHTTP] *can't send
[AHTTP] open( GET , url = http://192.168.1.21/ping.htm
[AHTTP] _parseURL(): scheme+host HTTP:// 192.168.1.21
[AHTTP] _parseURL(): port+path+query 80 /ping.htm 
[AHTTP] open: conneting to hostname = 192.168.1.21:80
[AHTTP] _connect()
[AHTTP] send()
[AHTTP] _buildRequest()
[AHTTP] _HTTPmethod = 0
[AHTTP] GET  /ping.htm   HTTP/1.1

[AHTTP] host : 192.168.1.21:80 

[AHTTP] _send(), _request->available = 50
[AHTTP] *can't send
H[AHTTP] open( GET , url = http://192.168.1.21/ping.htm
[AHTTP] _parseURL(): scheme+host HTTP:// 192.168.1.21
[AHTTP] _parseURL(): port+path+query 80 /ping.htm 
[AHTTP] open: conneting to hostname = 192.168.1.21:80
[AHTTP] _connect()
[AHTTP] send()
[AHTTP] _buildRequest()
[AHTTP] _HTTPmethod = 0
[AHTTP] GET  /ping.htm   HTTP/1.1

[AHTTP] host : 192.168.1.21:80 

[AHTTP] _send(), _request->available = 50
[AHTTP] *can't send
H[AHTTP] open( GET , url = http://192.168.1.21/ping.htm
[AHTTP] _parseURL(): scheme+host HTTP:// 192.168.1.21
[AHTTP] _parseURL(): port+path+query 80 /ping.htm 
[AHTTP] open: conneting to hostname = 192.168.1.21:80
[AHTTP] _connect()
[AHTTP] send()
[AHTTP] _buildRequest()
[AHTTP] _HTTPmethod = 0
[AHTTP] GET  /ping.htm   HTTP/1.1

[AHTTP] host : 192.168.1.21:80 

[AHTTP] _send(), _request->available = 50
[AHTTP] *can't send
H[AHTTP] _onError handler error = -13
[AHTTP] 
_onDisconnect handler
[AHTTP] _setReadyState : 4

**************************************
[AHTTP] responseText()
[AHTTP] responseText() no data

**************************************

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Note:  It takes approximately 30 seconds from the first [AHTTP] *can't send till the _onDisconnect handler fires.  The reset happens at the time the next request would happen after the _onDisconnect handler fires.

Thanks for your help

_Originally posted by @andrewk123 in https://github.com/khoih-prog/AsyncHTTPRequest_Generic/issues/16#issuecomment-955784844_

khoih-prog, 2021-11-23:
Hi @andrewk123 

Can you post your `internal NodeJS server code`

khoih-prog, 2021-11-23:
Hi @andrewk123 

I couldn't duplicate the issue as the code was running normally. I suggest you check your board / WebServer code

The debug terminal

```

**************************************
[AHTTP] responseText()
[AHTTP] xbuf::readString: Reserved size =  612
[AHTTP] responseText(char) <!DOCTYPE html>
<html lang="en">
<head>
    <meta charset="UTF-8">
    <meta name="viewport" content="width=device-width, initial-scale=1.0">
    <meta http-equiv="X-UA-Compatible" content="ie=edge">
    <title>Document</title>
</head>
<body>
    <form action="/" method="POST">
        <label>Name: </label>
        <input type="text" name="dname" value="" /><br />
        <label>Email: </label>
        <input type="text" name="demail" value="" /><br />
        <label>Address: </label>
        <input type="text" name="daddress" value="" /><br />
        <button>submit</button>
    </form>
</body>
</html>
 , avail = 611
<!DOCTYPE html>
<html lang="en">
<head>
    <meta charset="UTF-8">
    <meta name="viewport" content="width=device-width, initial-scale=1.0">
    <meta http-equiv="X-UA-Compatible" content="ie=edge">
    <title>Document</title>
</head>
<body>
    <form action="/" method="POST">
        <label>Name: </label>
        <input type="text" name="dname" value="" /><br />
        <label>Email: </label>
        <input type="text" name="demail" value="" /><br />
        <label>Address: </label>
        <input type="text" name="daddress" value="" /><br />
        <button>submit</button>
    </form>
</body>
</html>

************************************** <========== Stop server
[AHTTP] _onData handler 0

 , len = 5
[AHTTP] _onData: _processChunks
[AHTTP] _processChunks() 0

 , chunks available = 5
[AHTTP] xbuf::readString: Reserved size =  4
[AHTTP] *getChunkHeader 0
 , chunkHeader length = 3
[AHTTP] *all chunks received - no disconnect
[AHTTP] 
_onDisconnect handler
[AHTTP] open( GET , url = http://192.168.2.30:3000/index
[AHTTP] _parseURL(): scheme+host HTTP:// 192.168.2.30
[AHTTP] _parseURL(): port+path+query 3000 /index 
[... 380 lines cut ...]

2. index.html

```
<!DOCTYPE html>
<html lang="en">
<head>
    <meta charset="UTF-8">
    <meta name="viewport" content="width=device-width, initial-scale=1.0">
    <meta http-equiv="X-UA-Compatible" content="ie=edge">
    <title>Document</title>
</head>
<body>
    <form action="/" method="POST">
        <label>Name: </label>
        <input type="text" name="dname" value="" /><br />
        <label>Email: </label>
        <input type="text" name="demail" value="" /><br />
        <label>Address: </label>
        <input type="text" name="daddress" value="" /><br />
        <button>submit</button>
    </form>
</body>
</html>
```

andrewk123, 2021-11-23:
Thanks for your response.  

It actually does not matter what the node.js code is since the point is that the target ip is not available. 

Andrew

andrewk123, 2021-11-23:
Let me qualify my previous comment.  This error is caused when the target host is not available (ie there is no response from the host at all (such as the target host is completely down).  It is the lack of response that causes the issue (as described above).  Try this again with 

requestOpenResult = request.open("GET", "http://192.168.99.30:3000/index"); 

where 192.168.99.30 is a none existing IP address.

Please re-open this issue.

khoih-prog, 2021-11-23:
> Multiple requests to an internal NodeJS server when the server is **disconnected** causes a crash

> none existing IP address.

These 2 are different scenarios, but anyway, I can duplicate the **crash** only with `none existing IP address` now and will investigate soon.

Certainly, please post anything you've investigated / experienced so far to help speed up the tests / fix. I'm opening the issue now.

khoih-prog, 2021-11-23:
OK, found the bug. I'll publish a new release by tomorrow.

khoih-prog, 2021-11-23:
Hi @andrewk123 

The [**AsyncHTTPRequest_Generic releases v1.4.0**](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/releases/tag/v1.4.0) has just been published.

Your contribution is noted in [**Contributions and Thanks**](https://github.com/khoih-prog/AsyncHTTPRequest_Generic#contributions-and-thanks).

Please have more tests to verify the bug has been squashed.

Regards,

---

### Releases v1.4.0

1. Fix crashing bug when request a non-existing IP. Check [**Http GET polling causes crash when host disconnected** #22](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/issues/22)
2. Modify `platform.ini` to avoid compile error with PIO when using ESP8266/ESP32

closed 2021-11-23 by khoih-prog

andrewk123, 2021-11-23:
Thanks!  That seems to have completely resolved the issue.  I will do some further testing this evening and advise if I find anything of interest.  Thanks for your work on this!