khoih-prog/AsyncHTTPRequest_Generic #25: Compile error with ESP8266 and Platformio

Phunkafizer, 2021-11-14:
### Describe the bug

I created a new project in platformio (Win10)  (platform ESP8266 board D1) and added the library to the project. When compiling I get the following error:

compiling .pio\build\d1\lib6a0\STM32Ethernet\Dhcp.cpp.o
Compiling .pio\build\d1\lib6a0\STM32Ethernet\Dns.cpp.o
Compiling .pio\build\d1\lib6a0\STM32Ethernet\EthernetClient.cpp.o
Compiling .pio\build\d1\lib6a0\STM32Ethernet\EthernetServer.cpp.o
In file included from .pio\libdeps\d1\STM32Ethernet\src/lwipopts.h:17:0,
                 from .pio\libdeps\d1\STM32duino LwIP\src/lwip/opt.h:51,
                 from .pio\libdeps\d1\STM32duino LwIP\src/lwip/init.h:40,
                 from <userdir>\platformio\packages\framework-arduinoespressif8266@3.20704.0\cores\esp8266/IPAddress.h:27,
                 from .pio\libdeps\d1\STM32Ethernet\src\STM32Ethernet.h:5,
                 from .pio\libdeps\d1\STM32Ethernet\src\EthernetServer.cpp:5:
.pio\libdeps\d1\STM32Ethernet\src/lwipopts_default.h:89:0: warning: "TCP_MSS" redefined [enabled by default]
 #define TCP_MSS                 (1500 - 40)   /* TCP_MSS = (Ethernet MTU - IP header size - TCP header size) */
 ^
<command-line>:0:0: note: this is the location of the previous definition
In file included from .pio\libdeps\d1\STM32Ethernet\src/lwipopts.h:17:0,
                 from .pio\libdeps\d1\STM32duino LwIP\src/lwip/opt.h:51,
                 from .pio\libdeps\d1\STM32duino LwIP\src/lwip/init.h:40,
                 from <userdir>platformio\packages\framework-arduinoespressif8266@3.20704.0\cores\esp8266/IPAddress.h:27,
                 from <userdir>platformio\packages\framework-arduinoespressif8266@3.20704.0\cores\esp8266/Udp.h:39,
                 from .pio\libdeps\d1\STM32Ethernet\src\EthernetUdp.h:40,
                 from .pio\libdeps\d1\STM32Ethernet\src\Dns.cpp:5:
.pio\libdeps\d1\STM32Ethernet\src/lwipopts_default.h:89:0: warning: "TCP_MSS" redefined [enabled by default]
 #define TCP_MSS                 (1500 - 40)   /* TCP_MSS = (Ethernet MTU - IP header size - TCP header size) */
 ^
<command-line>:0:0: note: this is the location of the previous definition
In file included from .pio\libdeps\d1\STM32Ethernet\src/lwipopts.h:17:0,
                 from .pio\libdeps\d1\STM32duino LwIP\src/lwip/opt.h:51,
                 from .pio\libdeps\d1\STM32duino LwIP\src/lwip/init.h:40,
                 from <userdir>.platformio\packages\framework-arduinoespressif8266@3.20704.0\cores\esp8266/IPAddress.h:27,
                 from <userdir>.platformio\packages\framework-arduinoespressif8266@3.20704.0\cores\esp8266/Udp.h:39,
                 from .pio\libdeps\d1\STM32Ethernet\src\EthernetUdp.h:40,
                 from .pio\libdeps\d1\STM32Ethernet\src\Dhcp.h:7,
                 from .pio\libdeps\d1\STM32Ethernet\src\Dhcp.cpp:6:
.pio\libdeps\d1\STM32Ethernet\src/lwipopts_default.h:89:0: warning: "TCP_MSS" redefined [enabled by default]
 #define TCP_MSS                 (1500 - 40)   /* TCP_MSS = (Ethernet MTU - IP header size - TCP header size) */
 ^
<command-line>:0:0: note: this is the location of the previous definition
In file included from .pio\libdeps\d1\STM32Ethernet\src\EthernetUdp.h:43:0,
                 from .pio\libdeps\d1\STM32Ethernet\src\Dhcp.h:7,
                 from .pio\libdeps\d1\STM32Ethernet\src\Dhcp.cpp:6:
.pio\libdeps\d1\STM32Ethernet\src\utility/stm32_eth.h:42:23: fatal error: stm32_def.h: No such file or directory

Kind regards
Stefan

khoih-prog, 2021-11-14:
Hi,

Thanks for your interest in the library.

As you know, using PIO requires a little more experiences and knowledge than Arduino IDE to understand and manually modify the settings. Moreover, the current PIO version is **not smart enough to select the correct library according the board** necessary to use with this library, which is written mainly for **multi-platforms** and Arduino IDE.

**Just try using Arduino IDE and understand why it can selects correctly the libraries to use, no matter boards / platforms you're using.**

This library's `platformio.ini` is also written to **include all possible dependent libraries for all the boards with the expectation that users will modify to ignore the incorrect ones**. It's better than include nothing and users has to do research to know which libraries are needed and include them.

Just understand which libraries to be used for your board, you'll know that you can **ignore some unused libraries**, such as adding to your `platformio.ini`

```
lib_ignore = 
   STM32AsyncTCP
   STM32duino LwIP
   STM32Ethernet
   WebServer_WT32_ETH01
```

Good Luck,

closed 2021-11-14 by khoih-prog

Phunkafizer, 2021-11-17:
Thank you so much, with the lib_ignore section it is working like a charme!