khoih-prog/AsyncHTTPRequest_Generic #34: Request will get stuck in readyStateOpened after 25 or more successful requests

caspersauce, 2022-07-22:
I have a ESP32-WROOM running this library using the Arduino IDE. This device pings a server every 5 minutes to pull down some new data. I followed an example that works great. The problem is that after anywhere between 10-100 good requests request.readyState() continually says it is in readyStateOpened. This hangs up all future calls and the application is stuck.

I sadly cannot consistently reproduce this, so I can't offer much more details.

My question is why would this happen in general? Is there not a timeout to force it out of this state if there is an issue? Is there anything I can call to completely reset the request module so that I can at least begin using it again? I have tried request.abort() and it doesn't affect the readyState.

Thanks for the help!

khoih-prog, 2022-07-23:
Hi @caspersauce 

Thanks for your interest and usage of this library.

I suggest that you first try the [**AsyncHTTPRequest_ESP**](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/tree/master/examples/AsyncHTTPRequest_ESP) or [**AsyncHTTPMultiRequests_ESP** example](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/tree/master/examples/AsyncHTTPMultiRequests_ESP) to see if there is the same problem.

If not, there must be something wrong in **your code, server, network, etc.**

I certainly wish I have time to investigate everything, especially this **vague** and lack-of-info issue.

As you're using **ESP32**, I suggest that you retest using [**asyncHTTPrequest** library](https://github.com/boblemaire/asyncHTTPrequest), if there is similar hang-up, post the issue there.

Also remember to post [**Minimal, Reproducible Example**](https://stackoverflow.com/help/minimal-reproducible-example) with more info, code, etc. as possible, so that anybody can reproduce the problem.

I'm closing this issue now and only reopen after a bug of this library is **proven and reproduced**.

Good Luck,

closed 2022-07-23 by khoih-prog

caspersauce, 2022-07-23:
I appreciate your response! I understand that my question was vague and unreproducible, that was why I asked a very specific question:

When the readyState variable is stuck in readyStateOpened how do you get it out of that state? 

Despite if it is "code, server or network" it would be useful to know how to reset the module when an external issue happens, and I don't see anything in the documentation. As I mentioned the abort function does not seem to do anything.

Thanks!

caspersauce, 2022-07-26:
Just as update for future users who have sporadic issues. As was recommended I changed to the asyncHTTPrequest library. This library works perfectly and I have yet to have an issue with the readyState variable getting stuck.

If you are using and ESP32 I highly recommend using that library.

khoih-prog, 2022-07-26:
HI @caspersauce 

Thanks for sharing the experience.

If having time, I'll definitely have a look to see what are the differences between the 2 libraries and fix this issue.

I'd appreciate it if you can create and post the [**Minimal, Reproducible Example**](https://stackoverflow.com/help/minimal-reproducible-example) so that **I can reopen this issue** then, fix it and verify if the ESP32-sporadic issues have been fixed.