khoih-prog/AsyncHTTPRequest_Generic #36: Redundant code

Colkerr, 2022-08-20:
### AsyncHTTPRequest_ESP_Multi/AsyncHTTPRequest_ESP_Multi.ino
It looks like the for loop is unnecessary as reqCount[..]  is subsequentely overwritten.

void sendRequests()    //NUM_DIFFERENT_SITES == 2
{
  for (int index = 0; index < NUM_DIFFERENT_SITES; index++)
  {
    reqCount[index] = 2;
  }
  reqCount[0] = NUM_ENTRIES_SITE_0;  //that is == 2
  reqCount[1] = NUM_ENTRIES_SITE_1;  //that is == 1
}

khoih-prog, 2022-08-20:
Thanks. 
Just updated in `master`

closed 2022-08-20 by khoih-prog

Colkerr, 2022-08-20:
> <khoih-prog/AsyncHTTPRequest_Generic/issue/36/issue_event/7226471978@
> github.com>

Wow my first ever issue - thank you for the code, it's working so well and
I am trying to learn from it. My blocking GET went from sometimes 10s to
always milliseconds !!

regards Colin