khoih-prog/AsyncHTTPRequest_Generic #37: Getting no response when making invalid-endpoint requests

miwied, 2022-08-26:
Here is my test code: https://gist.github.com/miwied/98c64c6a1f60ab90b46d12e555d8e813

When I am manually making http get requests to an invalid endpoint I get no error response.
The "requestOpenResult" boolean is true all the time, no matter if the endpoint is reachable or not.

miwied, 2022-08-26:
Sorry I forgot to put the `    Serial.println(request->responseHTTPcode());
` in my code.

Now I am getting response code, but weird ones. For example -13 or -44, what does this mean?

khoih-prog, 2022-08-26:
https://github.com/khoih-prog/AsyncHTTPRequest_Generic/blob/2b193f78d3f67784977fba81e302d947d70ec056/src/AsyncHTTPRequest_Generic.hpp#L211-L221

To get more debugging codes, use

```
// Level from 0-4
#define ASYNC_HTTP_DEBUG_PORT     Serial
#define _ASYNC_HTTP_LOGLEVEL_     4 
```

Sorry I don't have time to help with these **mundane** issues, you're on your own **to do more research** before doing something hastily.

I'm closing the issue now as it's not a bug of the library.

closed 2022-08-26 by khoih-prog