khoih-prog/AsyncHTTPRequest_Generic #38: Setting a timeout have no effect on invalid endpoint requests

miwied, 2022-08-27:
### Describe the bug

If address is not an ip address e.g.: "http://exampleasynchttp.com/index.html", it gives immediately a "-4" code (good).
But if the address is like "http://192.168.0.44/index.html" (this host is down) the timeout takes always about 18 second no matter how many seconds you set (not good).

### Steps to Reproduce

Copy code from: 
https://gist.github.com/miwied/f56aca82c4f976f448dde1a8da051650
Flash it to esp32 and run.

### Expected behavior

The request should time out after **3** seconds.

### Actual behavior

The request time out after **18** seconds (no matter what you pass into the setTimeout() method).

khoih-prog, 2022-08-31:
Hi @miwied 

The new [**AsyncHTTPRequest_Generic releases v1.9.0**](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/releases/tag/v1.9.0) has just been published. Your contribution is noted in [**Contributions and Thanks**](https://github.com/khoih-prog/AsyncHTTPRequest_Generic#contributions-and-thanks)

Best Regards,

---

### Releases v1.9.0

1. Fix long timeout if using `IPAddress`. Check [setTimeout() hasn't any effect #38](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/issues/38)
2. Optimize code
3. Display only successful responseText in examples
4. Improve debug messages by adding functions to display error messages instead of `cryptic error number`
5. Remove support to STM32 using **LAN8720** due to problem with new STM32 core v2.3.0
5. Update `Packages' Patches`

---

Use the following code to test the new library version

### Test Code

```
#include <Arduino.h>

#if (ESP8266)
  #include <ESP8266WiFi.h>
#elif (ESP32)
  #include <WiFi.h>
#endif

#define ASYNC_HTTP_DEBUG_PORT     Serial
#define _ASYNC_HTTP_LOGLEVEL_     1

// 300s = 5 minutes to not flooding
#define HTTP_REQUEST_INTERVAL     60  //300

#define USING_GENERIC         true
//#define USING_GENERIC         false

#if USING_GENERIC
  #include <AsyncHTTPRequest_Generic.h>     // https://github.com/khoih-prog/AsyncHTTPRequest_Generic
#else
  #include <asyncHTTPrequest.h> 
  #define AsyncHTTPRequest    asyncHTTPrequest

  #define readyStateUnsent    0
  #define readyStateDone      4
  
#endif

#include <Ticker.h>

const char *ssid = "your_ssid";
const char *password = "your_pass";

AsyncHTTPRequest request;
Ticker ticker;

void sendRequest()
{
  static bool requestOpenResult;

  if (request.readyState() == readyStateUnsent || request.readyState() == readyStateDone)
  {
    //requestOpenResult = request.open("GET", "http://192.168.178.62/cm?cmnd=Power");
    //requestOpenResult = request.open("GET", "http://exampleasynchttp.com/index.html");
    requestOpenResult = request.open("GET", "http://worldtimeapi.org/api/timezone/America/Toronto.txt");

[... 143 lines cut ...]
00:36:58.636 -> [AHTTP] Response Code =  NOT_CONNECTED
00:36:58.636 -> [AHTTP] Response error
00:37:16.878 -> [AHTTP] _onError handler error = Connection aborted    <==== Controlled by AsyncTCP library
```

This case, `Response error` right away. But `Connection aborted` is controlled by AsyncTCP library and **delay by 18s**

---

### Using bad hostname `http://exampleasynchttp.com/index.html` =>  DNS error

```
00:38:33.359 -> Starting AsyncHTTPRequest_Test using ESP32_DEV
00:38:33.359 -> AsyncHTTPRequest_Generic v1.9.0
00:38:33.492 -> Connecting to WiFi SSID: HueNet1
00:38:33.989 -> ...
00:38:34.982 -> AsyncHTTPRequest @ IP : 192.168.2.109
00:38:34.982 -> Send request
00:38:34.982 -> [AHTTP] **************************************
00:38:34.982 -> [AHTTP] Response Code =  NOT_CONNECTED
00:38:34.982 -> [AHTTP] Response error
00:38:35.148 -> [AHTTP] _onError handler error = DNS failed
```

This case, `Response error` right away. `DNS failed` is also controlled by AsyncTCP library and activated **immediately**

closed 2022-08-31 by khoih-prog

miwied, 2022-08-31:
Hi @khoih-prog, first of all I would like to thank you for addressing the problem so quickly and looking for a possible solution.

Unfortunately, I have to tell you that the results are different for me than for you and the problem still exists. 😕
- - -
Here is the code is used (your test code):
```
#include <Arduino.h>

#if (ESP8266)
#include <ESP8266WiFi.h>
#elif (ESP32)
#include <WiFi.h>
#endif

#define ASYNC_HTTP_DEBUG_PORT Serial
#define _ASYNC_HTTP_LOGLEVEL_ 4

// 300s = 5 minutes to not flooding
#define HTTP_REQUEST_INTERVAL 60 // 300

#define USING_GENERIC true
//#define USING_GENERIC         false

#if USING_GENERIC
#include <AsyncHTTPRequest_Generic.h> // https://github.com/khoih-prog/AsyncHTTPRequest_Generic
#else
#include <asyncHTTPrequest.h>
#define AsyncHTTPRequest asyncHTTPrequest

#define readyStateUnsent 0
#define readyStateDone 4

#endif

#include <Ticker.h>

const char *ssid = "###";
const char *password = "###";

AsyncHTTPRequest request;
Ticker ticker;

void sendRequest()
{
  static bool requestOpenResult;

  if (request.readyState() == readyStateUnsent || request.readyState() == readyStateDone)
  {
    requestOpenResult = request.open("GET", "http://192.168.178.62/cm?cmnd=Power");
    // requestOpenResult = request.open("GET", "http://exampleasynchttp.com/index.html");
    // requestOpenResult = request.open("GET", "http://worldtimeapi.org/api/timezone/America/Toronto.txt");

    if (requestOpenResult)
    {
      // Only send() if open() returns true, or crash
      Serial.println(F("Send request"));

      request.send();
    }
    else
    {
      Serial.println(F("Can't send bad request"));
    }
  }
  else
  {
    Serial.println(F("Can't send request"));
  }
}

[... 120 lines cut ...]
09:47:30.485 > [AHTTP]
09:47:30.486 > [AHTTP] GET  /index.html
09:47:30.488 > _onDisconnect handler
09:47:30.490 > HTTP/1.1
09:47:30.491 >
09:47:30.491 > [AHTTP] host : exampleasynchttp.com:80
09:47:30.495 >
09:47:30.495 > [AHTTP] _send(), _request->available = 58
09:47:30.499 > [AHTTP] !connected yet
09:47:30.501 > [AHTTP] HTTPCODE_NOT_CONNECTED
09:47:30.504 > [AHTTP] _onDisconnect: _HTTPcode = -4
09:47:30.507 > [AHTTP] _onDisconnect: _HTTPcode = NOT_CONNECTED
09:47:30.511 > [AHTTP] _setReadyState : 4
09:47:30.514 > [AHTTP] **************************************
09:47:30.518 > [AHTTP] Response Code =  NOT_CONNECTED
09:47:30.521 > [AHTTP] Response error
```
---

Look closely on the timestamps in both logs. If requesting the bad ip address the lines 
```
[AHTTP] Response Code =  NOT_CONNECTED
[AHTTP] Response error
```
are delayed by 18 seconds as well... this does not occur when requesten the bad hostname.

khoih-prog, 2022-08-31:
Please read pevious answer

### Using bad IPAddress `http://192.168.178.62/cm?cmnd=Power`

```
00:36:57.012 -> Starting AsyncHTTPRequest_Test using ESP32_DEV
00:36:57.012 -> AsyncHTTPRequest_Generic v1.9.0
00:36:57.112 -> Connecting to WiFi SSID: HueNet1
00:36:57.609 -> ...
00:36:58.636 -> AsyncHTTPRequest @ IP : 192.168.2.109
00:36:58.636 -> Send request
00:36:58.636 -> [AHTTP] **************************************
00:36:58.636 -> [AHTTP] Response Code =  NOT_CONNECTED
00:36:58.636 -> [AHTTP] Response error
00:37:16.878 -> [AHTTP] _onError handler error = Connection aborted    <==== Controlled by AsyncTCP library
```

This case, **`Response error` right away**. But `Connection aborted` is controlled by AsyncTCP library and **delay by 18s**

---

Sorry I won't spend any more time on this special / not-popular use case.
I'm sure the original [**asyncHTTPrequest** library](https://github.com/boblemaire/asyncHTTPrequest) has the same issue

As you're using ESP32, post the issue there if you're still interested.

miwied, 2022-08-31:
@khoih-prog Ok. In my case the `Response error` unfortunately does not comes right away.. it is also delayed by 18 seconds. I thought you maybe know why. Thanks anyways ☹️