khoih-prog/AsyncHTTPRequest_Generic #43: Callback behaviour is buggy (ESP8266)

dirkvranckaert, 2022-10-21:
I'm on the latest version of the library.
My scenario: every 10 seconds I want to make a POST call to a server containing a JSON body.

The full testing code is available here: https://gist.github.com/dirkvranckaert/93ca29597a9f6732465d23cb9e2b62b3

Here is the log of what I observe:
```
11:41:06.936 -> ****************************************
11:41:07.010 -> ****************************************
11:41:07.010 -> ****************************************
11:41:07.087 -> 
11:41:07.087 -> 
11:41:07.087 -> Sending new request...
11:41:07.087 -> Network result... 
11:41:07.125 -> NOT_CONNECTED
11:41:07.125 -> -4
11:41:07.125 -> Network result callback done!
11:41:07.162 -> Network result... 
11:41:07.200 -> HTTP OK
11:41:07.200 -> 200
11:41:07.200 -> Network result callback done!
11:41:17.048 -> 
11:41:17.048 -> 
11:41:17.048 -> ****************************************
11:41:17.085 -> ****************************************
11:41:17.119 -> ****************************************
11:41:17.194 -> 
11:41:17.194 -> 
11:41:17.194 -> Sending new request...
11:41:17.339 -> Network result... 
11:41:17.339 -> Bad Request
11:41:17.378 -> 400
11:41:17.378 -> Network result callback done!
11:41:27.091 -> 

```

So my observation is:
1. The first network call is executed and reports twice in the callback, first with `HTTPCODE_NOT_CONNECTED` followed by the regular network response. That seems bizar to me that I first get the not connected followed by a proper response... I'm not sure if that's ok and if I can safely ignore the not connected...
2. Then we fire the second network call (same call just 10 minutes after), that call is started with success, however it immediately reports back with an HTTP error code 400 (I check my server and the request never arrived so it's definitely a client thing).

In my opinion both are an issue.

Can you please have a look?

dirkvranckaert, 2022-10-21:
Need to mention for testing that I created a simple PHP server with an empty index.php in it. That script I served with `php -S localhost:3000` from my computer and then used ngrok to give me an HTTP link to it.

dirkvranckaert, 2022-10-21:
I've investigated this some more from my side and it appears that every 2nd, 4th,... request (so after a 'good' request) no headers are being sent. My server shows me now that the Host is missing, the User-Agent is missing,...

khoih-prog, 2022-10-21:
HI @dirkvranckaert 

Please try the new [**AsyncHTTPRequest_Generic** v1.10.1](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/releases/tag/v1.10.1). Your contribution is noted in [**Contributions and Thanks**](https://github.com/khoih-prog/AsyncHTTPRequest_Generic#contributions-and-thanks)

---

### Releases v1.10.1

1. Fix bug of wrong `reqStates`. Check [**Release 1.9 breakes previously running code** #39](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/issues/39) and [**Callback behaviour is buggy (ESP8266)** #43](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/issues/43)
2. Optional larger `DEFAULT_RX_TIMEOUT` from default 3s, for slower networks

closed 2022-10-21 by khoih-prog

dirkvranckaert, 2022-10-21:
Great, I'll give it a try one of these days! Thank you for the quick update! Much appreciated!