khoih-prog/AsyncHTTPRequest_Generic #44: Host/Headers not always sent with 1.10.1

SeeliSoft-CH, 2022-11-03:
Hi
today I updated your library to 1.10.1 and now I spot an issue:

every second call seems to miss header informations, my server returns back:
`<h1>Bad Message 400</h1><pre>reason: No Host</pre>`

User @dirkvranckaert was already mentioning the same on issue #43, but I can't see if he was having that issue with 1.10.1 or 1.9.1. I did not have the issue with 1.9.1 (but the buggy callback behaviour instead). 
https://github.com/khoih-prog/AsyncHTTPRequest_Generic/issues/43#issuecomment-1286780712

Unfortunately I struggle to enable debugging, possibly it can be reproduced with the example provided with #43.

I use an ESP32 and the endpoint is an IP-address (no DNS), just in case that might matter.

khoih-prog, 2022-11-05:
Hi @SeeliSoft-CH 

Thanks for reporting the issue.

But sorry that no one can do anything, until you provide the complete [**MRE**](https://stackoverflow.com/help/minimal-reproducible-example) so that we can duplicate the issue.

Please do more tests to prove this is actually a bug of the library or just in your use-case (code /  server, etc.)

BR,

SeeliSoft-CH, 2022-11-06:
Hi @khoih-prog
I created a sample project to illustrate the problem.
[Bug-Demo.txt](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/files/9944935/Bug-Demo.txt)
[Serial Monitor Output.txt](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/files/9944936/Serial.Monitor.Output.txt)

In this sample the call is performed at every 5 seconds. Every second call returns in a 400 bad request error (as some header details are missing).
What I can already take ahead: the issue exactly always happens when it prints "already connected".

If I change the intervall to 10 seconds, the issue won't appear, as it will always open a new connection.

`[AHTTP] open( GET , url = http://www.seelisoft.net/test.html
[AHTTP] open: connecting to hostname = www.seelisoft.net:80
HTTP code: 200
`

`[AHTTP] open( GET , url = http://www.seelisoft.net/test.html
[AHTTP] open: already connected
HTTP code: 400
`

Hope that helps.

khoih-prog, 2022-11-10:
HI @SeeliSoft-CH 

Sorry I couldn't duplicate the issue using the example with 5s request interval, even with more complex site, Check the debug terminal with timestamps.

```
http://worldtimeapi.org/api/timezone/America/Toronto.txt
```

Possibly something wrong ( not compatible / slow network , etc.) of your `http` site. Try to investigate and fix.

Good Luck,

#### Terminal

```
22:15:09.158 -> Starting AsyncHTTPRequest_ESP using ESP32_DEV
22:15:09.158 -> AsyncHTTPRequest_Generic v1.10.1
22:15:09.257 -> Connecting to WiFi SSID: HueNet1
22:15:09.755 -> .......
22:15:12.775 -> AsyncHTTPRequest @ IP : 192.168.2.80
22:15:12.775 -> [AHTTP] open( GET , url = http://worldtimeapi.org/api/timezone/America/Toronto.txt
22:15:12.775 -> [AHTTP] open: connecting to hostname = worldtimeapi.org:80
22:15:12.775 -> [AHTTP] _client->connecting to worldtimeapi.org , 80
22:15:12.775 -> [AHTTP] client.connect OK to worldtimeapi.org , 80
22:15:13.073 -> [AHTTP] 
22:15:13.073 -> **************************************
22:15:13.073 -> [AHTTP] Response Code =  HTTP OK
22:15:13.073 -> 
22:15:13.073 -> **************************************
22:15:13.073 -> abbreviation: EST
22:15:13.073 -> client_ip: 65.110.220.219
22:15:13.073 -> datetime: 2022-11-09T22:15:12.992905-05:00
22:15:13.073 -> day_of_week: 3
22:15:13.073 -> day_of_year: 313
22:15:13.073 -> dst: false
22:15:13.073 -> dst_from: 
22:15:13.107 -> dst_offset: 0
22:15:13.107 -> dst_until: 
22:15:13.107 -> raw_offset: -18000
22:15:13.107 -> timezone: America/Toronto
22:15:13.107 -> unixtime: 1668050112
22:15:13.107 -> utc_datetime: 2022-11-10T03:15:12.992905+00:00
22:15:13.107 -> utc_offset: -05:00
22:15:13.107 -> week_number: 45
22:15:13.107 -> **************************************
22:15:17.753 -> [AHTTP] open( GET , url = http://worldtimeapi.org/api/timezone/America/Toronto.txt
22:15:17.753 -> [AHTTP] open: already connected
22:15:18.085 -> [AHTTP] 
22:15:18.085 -> **************************************
22:15:18.085 -> [AHTTP] Response Code =  HTTP OK
22:15:18.085 -> 
22:15:18.085 -> **************************************
22:15:18.085 -> abbreviation: EST
22:15:18.085 -> client_ip: 65.110.220.219
22:15:18.085 -> datetime: 2022-11-09T22:15:17.987679-05:00
22:15:18.085 -> day_of_week: 3
22:15:18.085 -> day_of_year: 313
22:15:18.085 -> dst: false
22:15:18.085 -> dst_from: 
22:15:18.085 -> dst_offset: 0
22:15:18.085 -> dst_until: 
22:15:18.085 -> raw_offset: -18000
22:15:18.085 -> timezone: America/Toronto
22:15:18.085 -> unixtime: 1668050117
22:15:18.085 -> utc_datetime: 2022-11-10T03:15:17.987679+00:00
22:15:18.085 -> utc_offset: -05:00
22:15:18.085 -> week_number: 45
22:15:18.085 -> **************************************
22:15:22.762 -> [AHTTP] open( GET , url = http://worldtimeapi.org/api/timezone/America/Toronto.txt
22:15:22.762 -> [AHTTP] open: already connected
22:15:22.762 -> H[AHTTP] 
22:15:23.127 -> **************************************
22:15:23.127 -> [AHTTP] Response Code =  HTTP OK
22:15:23.127 -> 
22:15:23.127 -> **************************************
22:15:23.127 -> abbreviation: EST
22:15:23.127 -> client_ip: 65.110.220.219
22:15:23.127 -> datetime: 2022-11-09T22:15:23.038722-05:00
22:15:23.127 -> day_of_week: 3
22:15:23.127 -> day_of_year: 313
22:15:23.127 -> dst: false
22:15:23.127 -> dst_from: 
22:15:23.160 -> dst_offset: 0
22:15:23.160 -> dst_until: 
22:15:23.160 -> raw_offset: -18000
22:15:23.160 -> timezone: America/Toronto
22:15:23.160 -> unixtime: 1668050123
22:15:23.160 -> utc_datetime: 2022-11-10T03:15:23.038722+00:00
22:15:23.160 -> utc_offset: -05:00
22:15:23.160 -> week_number: 45
22:15:23.160 -> **************************************
22:15:27.768 -> [AHTTP] open( GET , url = http://worldtimeapi.org/api/timezone/America/Toronto.txt
22:15:27.768 -> [AHTTP] open: already connected
22:15:28.067 -> [AHTTP] 
22:15:28.067 -> **************************************
22:15:28.067 -> [AHTTP] Response Code =  HTTP OK
22:15:28.067 -> 
22:15:28.067 -> **************************************
22:15:28.067 -> abbreviation: EST
22:15:28.067 -> client_ip: 65.110.220.219
22:15:28.067 -> datetime: 2022-11-09T22:15:27.990358-05:00
22:15:28.067 -> day_of_week: 3
22:15:28.067 -> day_of_year: 313
22:15:28.067 -> dst: false
22:15:28.067 -> dst_from: 
22:15:28.100 -> dst_offset: 0
22:15:28.100 -> dst_until: 
22:15:28.100 -> raw_offset: -18000
22:15:28.100 -> timezone: America/Toronto
22:15:28.100 -> unixtime: 1668050127
22:15:28.100 -> utc_datetime: 2022-11-10T03:15:27.990358+00:00
22:15:28.100 -> utc_offset: -05:00
22:15:28.100 -> week_number: 45
22:15:28.100 -> **************************************

```

khoih-prog, 2022-11-10:
HI @SeeliSoft-CH 

Anyway,  I'll create and release a new version to **default to resend headers**, etc. after already connected, to avoid error with new Apache Webserver

The test is OK with 5s between request now, w/o `Bad Requests`

```
23:11:33.395 -> Starting AsyncHTTPRequest_ESP using ESP32_DEV
23:11:33.395 -> AsyncHTTPRequest_Generic v1.10.2
23:11:33.494 -> Connecting to WiFi SSID: HueNet1
23:11:33.992 -> ...
23:11:35.020 -> AsyncHTTPRequest @ IP : 192.168.2.80
23:11:35.020 -> [AHTTP] open( GET , url = http://www.seelisoft.net/test.html
23:11:35.020 -> [AHTTP] open: connecting to hostname = www.seelisoft.net:80
23:11:35.020 -> [AHTTP] _client->connecting to www.seelisoft.net , 80
23:11:35.020 -> [AHTTP] client.connect OK to www.seelisoft.net , 80
23:11:35.451 -> [AHTTP] 
23:11:35.451 -> **************************************
23:11:35.451 -> [AHTTP] Response Code =  HTTP OK
23:11:35.451 -> 
23:11:35.451 -> **************************************
23:11:35.451 -> OK!
23:11:35.451 -> **************************************
23:11:40.029 -> [AHTTP] open( GET , url = http://www.seelisoft.net/test.html
23:11:40.029 -> [AHTTP] open: connecting to hostname = www.seelisoft.net:80
23:11:40.460 -> [AHTTP] 
23:11:40.460 -> **************************************
23:11:40.460 -> [AHTTP] Response Code =  HTTP OK
23:11:40.460 -> 
23:11:40.460 -> **************************************
23:11:40.460 -> OK!
23:11:40.460 -> **************************************
23:11:45.004 -> [AHTTP] open( GET , url = http://www.seelisoft.net/test.html
23:11:45.004 -> [AHTTP] open: connecting to hostname = www.seelisoft.net:80
23:11:45.004 -> H[AHTTP] 
23:11:45.468 -> **************************************
23:11:45.468 -> [AHTTP] Response Code =  HTTP OK
23:11:45.468 -> 
23:11:45.468 -> **************************************
23:11:45.468 -> OK!
23:11:45.468 -> **************************************
23:11:50.013 -> [AHTTP] open( GET , url = http://www.seelisoft.net/test.html
23:11:50.013 -> [AHTTP] open: connecting to hostname = www.seelisoft.net:80
23:11:50.511 -> [AHTTP] 
23:11:50.511 -> **************************************
23:11:50.511 -> [AHTTP] Response Code =  HTTP OK
23:11:50.511 -> 
23:11:50.511 -> **************************************
23:11:50.511 -> OK!
23:11:50.511 -> **************************************
23:11:55.020 -> [AHTTP] open( GET , url = http://www.seelisoft.net/test.html
23:11:55.020 -> [AHTTP] open: connecting to hostname = www.seelisoft.net:80
23:11:55.020 -> H[AHTTP] 
23:11:55.518 -> **************************************
23:11:55.518 -> [AHTTP] Response Code =  HTTP OK
23:11:55.518 -> 
23:11:55.518 -> **************************************
23:11:55.518 -> OK!
23:11:55.518 -> **************************************
23:12:00.028 -> [AHTTP] open( GET , url = http://www.seelisoft.net/test.html
23:12:00.028 -> [AHTTP] open: connecting to hostname = www.seelisoft.net:80
23:12:00.426 -> [AHTTP] 
23:12:00.426 -> **************************************
23:12:00.426 -> [AHTTP] Response Code =  HTTP OK
23:12:00.426 -> 
23:12:00.426 -> **************************************
23:12:00.426 -> OK!
23:12:00.426 -> **************************************

```

khoih-prog, 2022-11-10:
Hi @SeeliSoft-CH 

The new [**AsyncHTTPRequest_Generic** v1.10.2](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/releases/tag/v1.10.2) has just been published. Your contribution is noted in [**Contributions and Thanks**](https://github.com/khoih-prog/AsyncHTTPRequest_Generic#contributions-and-thanks)

Please test and report more bugs.

Best Regards,

---

### Releases v1.10.2

1. Default to reconnect to the same `host:port` after connected for new HTTP sites. Check [**Host/Headers not always sent with 1.10.1** #44](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/issues44)
2. Update `Packages' Patches`

closed 2022-11-10 by khoih-prog

SeeliSoft-CH, 2022-11-11:
Many many thanks! I can confirm it's working fine.